OpenRepeater/openrepeater #8: RPI 2 Invalid value for Card

dwebber1, 2015-08-25:
Hi Guys awesome project ! I have a little issue of the RPI img. When trying to use USB audio card you recommend (http://www.amazon.com/gp/product/B007HISGRW?ref_=zg_bs_3015427011_11), and I use the defualt settings on the hardware config page "alsa:plughw:0 I get this error "ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
" Is there some extra config I have to do?
Thanks

alleghenycreative, 2015-08-26:
Hi @dwebber1,
Thanks for your interest in the project. I have different values on the same USB card when used on my RPI2 than I do on my BBB. Try logging in via SSH and running "aplay -l" from the GUI. This should show your USB card. Take the number in the  listed and sub that out for the number in the config page.

So....
"Card 0" = alsa:plughw:0
"Card 1" = alsa:plughw:1

Depending on your card on RX channel, 0 is probably best, this should be the tip on the mic jack. Should work for mono or stereo mic input.

For TX channel, use 0 for Left (tip) and 1 for Right (ring)

After you change your GPIO pins in the ports tab, make sure to save, then rebuild, then reboot. 

You can also use "service svxlink stop" and "service svxlink start" to restart the svxlink service

Hope this helps.

dwebber1, 2015-08-26:
Okay,
So from what I understand is that ONE USB audio card 
Has two addresses 
alsa:plughw:0 can be the RX, and 
alsa:plughw:1 can be the TX.

Currently my setup both have alsa:plughw:0, the green light flashes on the card, but it seems like the transmit is stuck open. My setup is the easy digi, uv-5r, and pi 2.
Is the problem that it is stuck down is because that my addresses are conflicting.
I am interesting in being involved in this project to further it.

Thank you
Daniel Webber

alleghenycreative, 2015-08-26:
No you are going to use "alsa:plughw:0" for both RX Audio Device and TX Audio Device, and if that doesn't work use "alsa:plughw:1" for both. Channel should be set to 0 on both as that should be mono/left for both in an out channels. 

I was just saying that my Beaglebone Black sees the card as alsa:plughw:1 where my RPI2 sees it as alsa:plughw:0

Like I mentioned above when you do "aplay -l":
"Card 0" = alsa:plughw:0
"Card 1" = alsa:plughw:1

Hope that makes sense. We plan to make a auto-detect feature that reads and parses the aplay results in a future release.

dwebber1, 2015-08-26:
Oh okay, thank you.When you mean channel should be set to zero, what are exactly talking about 
Thank you
Daniel

alleghenycreative, 2015-08-26:
In other word your Port Settings are going to look like this....

RX Audio Device: alsa:plughw:0
RX Audio Channel: 0
TX Audio Device: alsa:plughw:0
TX Audio Channel: 0

OR This...

RX Audio Device: alsa:plughw:1
RX Audio Channel: 0
TX Audio Device: alsa:plughw:1
TX Audio Channel: 0

dwebber1, 2015-08-26:
Okay thank you, 
Daniel Webber

dwebber1, 2015-08-26:
So this setup should work for
Simplex. On your setup page it says the RX channel is 0 and the TX channel is 1.
Daniel

alleghenycreative, 2015-08-26:
There is no simplex logic option setup at this point (ie and echolink node). Just full duplex repeater logic. May add that later. So you still need to receive at the same time you are transmitting. Channels 0 and 1 are left and right channels of the sound card. You only need one or the other.

alleghenycreative, 2015-08-26:
Also Future support issues would be best posted on the OpenRepeater forums on our website. 

http://openrepeater.com/forums

closed 2015-08-26 by alleghenycreative