OpenRepeater/openrepeater #25: ID does not output when timer is set to under 1 minute

Dloranger, 2017-04-29:
GUI accepts timer settings under 1 minute (fractional minutes).  When this value is less than 1 minute the ID never transmits.  suggest fixing the GUI to only accept whole minute values

bobruddy, 2019-02-08:
is this still an issue?

Dloranger, 2019-02-08:
Yes, its a php bug. Timer uses whole minutes in svxlink and must be 0 for disable or 1+ for actual use.

Should be a simple fixup in the gui

alleghenycreative, 2019-06-10:
Not really a bug in PHP. Just need some UI refinements like a JS/jQuery validation to check the values client side. This script could also check both values for short and long IDs and make sure that the it is multiple of the other.

"The LONG_IDENT_INTERVAL must be an even multiple of the SHORT_IDENT_INTERVAL so if LONG_IDENT_INTERVAL is 60 then the legal values for SHORT_IDENT_INTERVAL are: 1, 2, 3, 4, 5, 6, 10, 12, 15, 20, 30, 60. "