OpenRepeater/openrepeater #27: GUI Feature Request

Dloranger, 2017-04-29:
Dashboard should have an ORP & SVXlink build date or other revision info displayed

alleghenycreative, 2018-07-09:
There is the version number listed in the footer, is this not adequate enough? Let me know your opinion.

Dloranger, 2018-07-09:
you have orp version, but not svxlink version. suggest adding the svxlink
version as well. DB could be updated by the install script with either a
release or build date?

alleghenycreative, 2019-06-10:
While it wouldn't be too difficult implement putting the SVXLink release number into the database during the build process, that leaves no way to update it should the user make modifications to the underlying svxlink build. While less likely, I don't know if there is way to ascertain this info from somewhere in SVXLink that way the info would be dynamic and more relevant. Not a high priority item, but would be nice to include at some point.

Dloranger, 2024-04-09:
Chris is writing a method for pulling the svxlink version details, we should be able to get it from him and migrate to the orp_helper utility