OpenRepeater/openrepeater #42: Internal error 500 after apt upgrade

dterweij, 2018-01-13:
When you do apt update after installed the image, then its all kind of messed up.

- All /usr/bin/openrepeater_* files are gone
- Webinterface has internal error 500
- GPIO pins not working anymore (needed to do echo /sys/gpio/ things to get the PTT working and added 
 a ! to PTT_PIN=!gpio18 in svxlink.conf)
- Starting now svxlink manualy from cli
- Webinterface is not useable when using Ddr/Rtlusb as RX.

Now, I can go back to the image, but then I can not update the system with security packages.

Hope you can fix the /usr/bin/openrepeater_* scripts back and get rid of the 500 internal error.

alleghenycreative, 2018-01-15:
Hello Danny, Thanks for the feedback. I will try to take a look into this. I don't have a time frame for it however. I have taken a hiatus from the project from a while due to work demands this past year. I am hoping to start development improvements again soon.

dterweij, 2018-01-15:
No problem, take your time :)  The openrepeater.db was empty after the update. Replaced back the distributed openrepeater.db. Though the setup wizzard is not coming far now,  it does not see the audio interfaces anymore. So more is messed up :) For now I edit the svxlink config files manualy.  Still testing and playing arround. 
Also I builded the easy digi myself, but the PTT optocoupler is not working so I used a CD4066BE chip for that and its working fine. It seems something wrong with the easy digi optocoupler PTT schematic, might be a wrong resistor value. As there are many values found in schematics found on google. If i am right in the math, I have 150mA at the optocoupler input, way too much. The CD4066BE is cheaper and easier to use, so i remove teh optocoupler line and put the CD4066BE in place. The audio line is fine (just need one as my RX is from the RTL USB dongle).
So you might can add as feature request, detect the RTL USB Dongle and make it selectable in the webinterface with all the options needed (CTCSS/FQ correction, frequency, SQL detection mode etc etc).

But no hurry :)

alleghenycreative, 2018-07-09:
Hello Danny, I am marking this as not fixing and closing. The reason for this is because the new releases are not going to be packaged based anymore. This requires someone to maintain both packages for svxlink and orp which adds another level of complexity onto the project...so the "apt update" will no longer apply and will not put the orp install at risk when updating the rest of the system...ie security updates. So in some respects it will be fix going forward....just not in the expected way.

closed 2018-07-09 by alleghenycreative