OpenRepeater/openrepeater #43: for ICS-CTRL hardware alsamixer has wrong settings

Dloranger, 2018-07-12:
need to configure capture as line-in (default is mic) and set the volume levels

alleghenycreative, 2018-07-13:
Per our discussion via slack DM for others to comment:

It might be something where we need to add the script function (to orp_helper script) that interfaces with alsa, then add the settings as part of the board preset array. When a board is selected, then it will pass those settings into alsa if they exist. 

This will allow us to have different alsa settings for different boards which could potentially have different audio codecs.

alleghenycreative, 2018-07-13:
Next step on this is to figure out how to script the ALSA Mixer in bash.

Dloranger, 2018-07-28:
amixer set 'Headphone' 86%; amixer set 'PCM' 75%; amixer set 'Lineout' 58%; amixer set 'Mic' 59%; amixer set 'Capture Mux' LINE_IN;

alleghenycreative, 2018-07-30:
Thanks for the feedback on the values for the ICS boards using the Fe-Pi audio. 

FOR OTHERS: Dan and I have been testing some initial command line code that we've exchanged outside of this issue. See his comment above for an example of this code with updated values. This will set the mixer settings...assuming that the Fe-Pi is the default and works well.

Next: We will need to continue testing the command line portion. There have been a few issues where the desired card is not the default or card 0. I am no Alsa expert, but from what I've read it has been inferred that you can set this as well which would need to be done before issuing the other level commands. My understanding is that is done by device number: however, we will need to figure out how to do by device name. This may be we need to grep aplay/arecord for the number based on the name and store the card number into a variable and then set it by card number. 

The end goal is to add the card name (i.e. Fe-Pi) as it appears to Debian and all the appropriate values into a PHP settings array (or later an XML file) and be able to pass those to a bash script (i.e. orp_helper) to set. Still some more testing do.

alleghenycreative, 2018-09-02:
This has been added into 2.0.0 code. Settings are stored in board presets file, and passed into alsamixer via the orp_helper script.

closed 2018-09-02 by alleghenycreative