OpenRepeater/openrepeater #54: Add support for higher channel count sound cards

Dloranger, 2018-11-27:
We need to expand beyond the base 2 channels for high definition sound cards

arecord -vvv -f dat /dev/null

will provide an output similar to the following (needs killed though as it runs forever), card zero (pi sound plug) doesn't seem to be named, but is the first in the list followed by the h

root@PI-REPEATER-4X:~# arecord -vvv -f dat /dev/null
Recording WAVE '/dev/null' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Route conversion PCM (sformat=S32_LE)
  Transformation table:
    0 <- 0
    1 <- 1
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Slave: Hardware PCM **card 1** '**seeed-8mic-voicecard**' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  **channels     : 8**
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
  appl_ptr     : 0
  hw_ptr       : 0

Dloranger, 2019-11-13:
Expanding this to be closer to the final solution, this command will read back the number of channels per card starting at card 0 (the on board sound device):

(arecord -vvv -f dat /dev/null --duration=1 --device=plughw:1,0) 2> >((grep "channels") > InputChannelsTMP.txt)

the file contents will be
  channels     : 2
  channels     : 10

This result could then be indexed using a regular expression search looking for a number, which could then be iterated into the populated list in the GUI.

Respectively for the playout channels, you can get the channel count by using the command
(aplay -vvv --device plughw:1,0 /usr/share/sounds/alsa/Side_Left.wav --duration 1) 2> >((grep "channels") > OutputChannelTMP.txt).

Discussion: the card zero numbers looks to be reversed, with expected 2 outputs and 1 input channels.  For the card 1 in this case, I can't tell as they are the same.   The outputs should be compared against a known card to ensure the lists are not transposed  input <-> output.

Dloranger, 2019-11-13:
according to speaker-test, here are the sound channel mappings for 8 channels
 0 - Front Left
 6 - Front Left Center
 2 - Front Center
 7 - Front Right Center
 1 - Front Right
 5 - Rear Right
 4 - Rear Left
 3 - LFE

Here are for 10 channels
 0 - Front Left
 6 - Front Left Center
 2 - Front Center
 7 - Front Right Center
 1 - Front Right
 5 - Rear Right
 8 - Rear Center
 4 - Rear Left
 9 - Side Left
 3 - LFE

alleghenycreative, 2019-11-14:
On the right track, but I think we will need to figure out how to refine this. The following does not work while the card is active: (arecord -vvv -f dat /dev/null --duration=1 --device=plughw:1,0) 2> >((grep "channels") > InputChannelsTMP.txt)

As it is trying to play/recored a sound, it needs access to the card but in some cases SVXLink will already be using the card, so we will need to be able to query the device passively. Maybe a different command? Also we will probably not want to write it to a file but just echo out the results as it will need to pass back into PHP for processing and writhing out the config variables. I will need to test how different cards behave with each other as well and it that causes any issues. 

Will probably also need to get the card numbers before passing this command or create some kind of loop to iterate through each card.

Dloranger, 2022-09-12:
improved command to interrogate the sound cards

$ ((arecord -vvv -f dat /dev/null --duration=1 --device=plughw:0,0) 2> >( grep -Pa  "channels"))| grep -o '[0-9][0-9]*'  2

limit is still that it targets a specific plughw:#,0 which is not ideal and needs to be expanded/refined to read all such devices

abcrawford, 2022-09-16:
I think that in theory this should be able to be used in conjunction with the script already in place to get the sound devices. This could be added to the orp_helper as an additional function. The device number could be passed in as an argument. 

Then on the PHP side the class will first get all the sound devices with the existing function and populate a PHP array, which is already does. Then it will iterate through the array and query each device for its number of channels and update the array accordingly. Currently the channels are hard coded, so this should in theory fix that and give a more realistic channel count…especially for high channel count devices. 

I will try to start testing this on my end. It think it will make send to develop that bash/orp_helper functions first. Then we can focus on the PHP side.

Dloranger, 2022-09-28:
Resolved with pull request

https://github.com/OpenRepeater/openrepeater/commit/e328a2661a3b0b06e9bc36774ab0bf9570d46415

closed 2022-09-28 by Dloranger