OpenRepeater/openrepeater #58: Ident page doesn't support the SVXLink IDENT_ONLY_AFTER_TX parameter

mytechguyri, 2019-01-23:
This additional parameter is an absolute MUST HAVE item.  A repeater continually identifying itself every 10 minutes 24/7 even when not being used is annoying, and considered downright rude in some areas. (FCC has actually cited it as anti-social behavior in some enforcement letters from what I understand).   While not very well documented in the SVXLink documentation, it appears that this parameter makes the Ident much more polite, in that, if the repeater is being actively used, the system will ident at the appropriate short ident time..... But if it is not being actively used, the repeater will remain silent.    Most examples of this parameter that I've seen have been IDENT_ONLY_AFTER_TX=4 (as I said, there doesn't appear to be any documentation on what different values do for the parameter.  but some digging through the support forums, I think I found a pretty good explanation.

"the number of seconds to wait after the last identification
was sent before we consider a transmission as "something new" for which we
need to ID again later, when the ID timer expires (to be legal).

In other words, if an ID is sent by identify, we won't trigger another ID
to be sent (probably 10 minutes later) unless it happens after the number
of seconds defined in IDENT_ONLY_AFTER_TX.  This lets us account for the
squelch tail or any other runs through transmit that are really part of
the identification transmission.")

But its clear that this needs to be an option field on the ident screen, so that users can set this parameter if they choose, and in so doing make their repeater less annoying, and more polite in its ident practices.

From SVXLink svxlink.conf man page.

IDENT_ONLY_AFTER_TX

This feature controls when identification is done. By default, identification is done every time the SHORT_IDENT_INTERVAL expires. If this feature is enabled, identification will be done only if there has been a recent transmission. This feature is good for nodes using an RF link to provide echolink to a repeater. Often, in this situation, it is not desirable for the link to identify unless legally necessary. Note that SHORT_IDENT_INTERVAL still have to be set for this feature to work. That config variable (SHORT_IDENT_INTERVAL) will then be interpreted as the minimum number of seconds between identifications. The LONG_IDENT_INTERVAL will not be affected by this parameter.

hayden-t, 2021-08-15:
https://openrepeater.com/knowledgebase/topic/custom-svxlink-configuration @David if you look there, near the end there is provision for adding custom svxlink config settings and not having them overwritten by openrepeater.

N2ZPY, 2025-10-26:
Have any changes been made on this issue?  I have 2 ORP ready to deploy but will not put them on the air until I can make it a polite system.

hayden-t, 2025-10-27:
you can set the setting manually for svxlink config