OpenRepeater/openrepeater #60: Dashboard is slow to load, 2.1.x beta

Dloranger, 2019-02-08:
Everything works but something is slowing down the page loading up. Likely an update routine can be isolated and optimized

alleghenycreative, 2019-02-08:
Seems 'includes/js/sys_info_chart_data.php' is taking about 5-6 seconds to load on 3B+

alleghenycreative, 2019-04-30:
Update on this issue. I was trying to investigate the lag issue…especially with the dashboard. On my latest build on the 3B+ it seems to be related to any files that are making calls to the Linux system. I clicked through all the pages and most are snappy. Obviously, the dashboard has lag (which is apparent when logging in) or going back to the dashboard. But also the Interface page, which displays system info in the “advanced details” popup. Both of these pages make calls to the Linux sub system through a PHP class and then the orp_helper script.

Trying to think of a solution. Either the class and scripts will need optimized or maybe I will need to load the info via AJAX only that way the page can load more promptly. I would need to display a spinner or something until the info gets loaded in via AJAX.Might take some rewriting both pages.