OpenRepeater/openrepeater #65: CTCSS Configuration not implemented correctly

mytechguyri, 2019-03-04:
It appears that the software CTCSS implementation (instead of using CTCSS on the actual radios, having SVXLINK decode it) is not implemented correctly.   In order to implement CTCSS detect, the following parameters need to be set

OPEN_ON_CTCSS=[tone in mhz]:[time_in_ms]
and also.... Since in this case, a user would run open squelch on their actual radio, you need to change
OPEN_ON_SQL={same as timout timer or a rediculously high amount of time 9999999999 ]..... Otherwise, repeater will open if the squelch is open for however long this value is set, regardless of CTCSS received.... so for example...

If 

OPEN_ON_CTCSS=67:100   (67hz CTCSS tone for 1 second)
and
OPEN_ON_SQL=1

Since the radio is open squelch, just keying the mic on the frequency for 1 second will open the repeater, even if the incorrect CTCSS tone, or no CTCSS tone at all is sent.   

By setting OPEN_ON_SQL=999999   The squelch would have to be held open for 1000 seconds with an incorrect CTCSS tone before it would open... It will still open within 100 ms if a valid CTCSS tone is present.

(In my case, I intend to run a dual Analog / DMR hybrid repeater.... Because of this, I need to run my radio with open squelch.... If the correct CTCSS tone is present, then we know its an analog signal, and ORP can open the repeater..... the MMDVM board will simply ignore the analog signal since there's no DMR color code present..    If a DMR signal is present, with the correct color code, since no CTCSS tone would be present, OpenRepeater should ignore the signal, but MMDVMHost would then open the repeater since it sees the digital signal with the correct color code.)

alleghenycreative, 2019-03-04:
@mytechguyri Interesting concept with the hybrid repeater setup. I've been watching the MMDVM project for a while and recently picked up a hotspot to start experimenting with with it. I've been considering including it into ORP in the future but that is a ways off and will likely require some different hardware designs. 

As for the software CTCSS decode/encode, that feature has always been kind of experimental and never fully tested. Many radios seem to high pass filter on the audio and not pass the received CTCSS tones on to the controller. For that reason we have up to currently recommended setting the CTCSS in the radios. 

Yes there is some room for improvement there and I can see your need for software CTCSS in what you are trying to achieve. This probably won't be addressed in 2.1.0 as it would be nice to get that released soon and not delay it any more that need be. Possibly 2.2.0. Maybe I can get you help with that. I would recommend testing on your end until after 2.1.0 is dropped. I will mark this as a feature request and we can continue the conversation.

mytechguyri, 2019-03-04:
The issue i opened here is more that orp isn't setting ctcss parameters
correctly..  Understood some radios may not support it and would filter it
out.... What I'm saying, if i set the two parameters i listed (configure it
correctly)   software ctcss works (not great, the detection is a bit hit or
miss, but it does work..).... Its a bug in the current setup that orp is
never setting the OPEN_ON_CTCSS variable...and is leaving OPEN_ON_SQL set
to 1 (which means even a wrong ctcss, if squelch is open for 1ms, it will
ignore ctcss and open the repeater.

If it's being viewed as a future feature, I'd completely remove the
software ctcss options from the web interface in 2.1.x because its not
setting the correct parameters in the config file, and so will never work
as it is..

--- Sent from my Android™ powered LG V20

mytechguyri, 2019-03-04:
> Assigned #65 <https://github.com/OpenRepeater/openrepeater/issues/65> to

Sometimes its easier to show than to type.

I think you'll see what i mean in the video at this link:

https://drive.google.com/file/d/1jKoiQdMnjONteNe10Mf0qshLnUoypagq/view?usp=drivesdk

--- Sent from my Android™ powered LG V20