OpenRepeater/openrepeater #68: Ider

KJ6IX, 2019-04-15:
Settings:
Short ID Settings
Morse ID
Time 0 - Min
Long ID Settings
60 - Min
During a QSO the voice ID's the repeater every 2 minutes. Looking at the config file this is not listed.
What I would like to do is change this 120 sec to 600 sec this would make it every 10 minutes. 
I'm new to this.
Dale KJ6IX

Dloranger, 2019-04-17:
i can confirm this bug, disabling the ids completely has the same effect

alleghenycreative, 2019-04-19:
Testing a fix for this now for the upcoming 2.1.0 image.

Dloranger, 2019-04-19:
@KJ6IX 

This looks to be a svxlink bug, and I have created a bug report there against it.
https://github.com/sm0svx/svxlink/issues/450

It has been chased down to the offending code, so we can now look at ways to fix/work-around it.  There are a handful of these hidden ID events in the code, so we may end up fixing several of them, but please know we are working on it, and hopefully we will get some insight from the svxlink github project members as well, its worth keeping an eye on the link to see if anything pops up there as well.

Dloranger, 2019-04-21:
I am working on scripting the newest fix into the installer script.

fix is commenting out the contents of file 
/usr/share/svxlink/events.d/RepeaterLogic.tcl, process "repeater_down" 

which has been found to be the source of the unwanted IDs through testing.

I am also working on fixing up a couple other small items before I push up

KJ6IX, 2019-04-21:
Thanks Dan

 

 

Dale Yanz  KJ6IX AFA9DY AFB9MC
 HM/CL:775-265-4954

  <mailto:KJ6IX@ARRL.NET> KJ6IX@ARRL.NET

 <mailto:AFA9DY@GMAIL.COM> AFA9DY@GMAIL.COM

 

“it is never too late to be prepared for the unexpected !” 

 

This email has been checked for viruses by AVG.
https://www.avg.com

alleghenycreative, 2019-04-30:
This should be corrected now. I have made a combination of some code updates in the PHP class, and @Dloranger and helped with with the 2.1.x build script change that will comment out the repeater_down and repeater_up procedures in the RepeaterLogic.TCL upon installation of SVXLink. This will be corrected in the official 2.1.0 build. Users can also correct this by editing their /usr/share/svxlink/events.d/RepeaterLogic.tcl file and commenting out the contents of the "proc repeater_down" and "proc repeater_up" functions. Then upon rebuild and restart the issue should go away.

closed 2019-04-30 by alleghenycreative