OpenRepeater/openrepeater #71: default login still present for use pi

Dloranger, 2019-05-18:
The pi user password needs updated or blocked so it will reduce the risk of hacking when hooked to an outside network

alleghenycreative, 2019-05-19:
I believe this may have caused and issue in the past. I think I disabled it in a past image, but I cannot remember what the issue was. I think in the setup guide I emphasis it's the end users responsibility to secure the Pi. Maybe we can add suggestions to a KB article. We can test with it disabled to make sure. Maybe add to the build script. Which would be unrelated to the "openrepeater" repo.

Dloranger, 2019-05-19:
I would suggest just making it match the root password so its not the
default raspbian password at least

alleghenycreative, 2019-05-19:
Maybe we can experiment with commands for the script. Probably better to move this issue over there.

closed 2019-05-19 by alleghenycreative