OpenRepeater/openrepeater #76: login.php fail

CR7AVR, 2019-07-21:
hi.
i´m new with this software.
i´m trying to install a simplex repeater.
the problem is wen i do the login, it says " 504 Gateway Time-out " and dont start the wizard...
i already did the configuration and expand the file system.
what can be wrong?
Please help

Best regards 
CR7AVR - Portugal

CR7AVR, 2019-07-22:
I have install the 2.0.0 version and now I runs fine... But not with 2.1.1 what could be the problem?

alleghenycreative, 2019-07-23:
Hello. I'm currently traveling so my ability to respond and test will be limited. I have not encountered this error myself. Are you trying to access through a ddns or some port forwarding or are you on the same network?

Have you also tried to download a second copy and burn on a different sd card?

Just need to rule things out first.

CR7AVR, 2019-07-23:
hi :)
i have the raspberry connected to the router inside the same network.
i just tried to make a image with another card and still the same.
i did the raspi-config like it´s explained in the website.
when i try to login as admin shows th e 504 error

alleghenycreative, 2019-07-23:
When you type in your IP into the web browser, do you get anything like the ORP logo or login page or does it go straight to the gateway error.

CR7AVR, 2019-07-23:
it shows the normal login page... the login.php .... when i do the login it goes to the 504 error page

i can provide a connection via teamviewer for anyone to try...

CR7AVR, 2019-07-23:
I wasked a friend to install in his raspberry and also have the same problem... :(

alleghenycreative, 2019-07-24:
Did you happen to update/upgrade the OS? We've experienced some weird behavior when doing so. That's why we've added a warning to the shell login. 

Perhaps even try a fresh burn without expanding the file system and see if that works. It will still give you 8GB which is plenty.

CR7AVR, 2019-07-24:
Hi :)
I tryed with expansion and without expansion. Upgrading and not upgrading the os... I have 3 different cards of 16gb each...

alleghenycreative, 2019-07-27:
Can you post the results of /var/log/nginx/error.log here

Do this after you've generated the issue so that it gets logged.

CR7AVR, 2019-07-28:
2019/05/17 02:23:07 [info] 11554#11554: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/17 02:30:59 [info] 27719#27719: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/17 02:51:09 [info] 559#559: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/18 02:52:08 [error] 653#653: *868 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.219, server: , request: "GET /functions/ajax_system.php?update=info HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.203", referrer: "https://192.168.1.203/dashboard.php"
2019/05/18 02:52:49 [error] 653#653: *888 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.219, server: , request: "GET /functions/ajax_system.php?update=info HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.203", referrer: "https://192.168.1.203/dashboard.php"
2019/05/18 02:53:31 [error] 653#653: *885 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.219, server: , request: "GET /functions/ajax_system.php?update=info HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.203", referrer: "https://192.168.1.203/dashboard.php"
2019/05/18 02:57:46 [error] 652#652: *947 FastCGI sent in stderr: "PHP message: PHP Warning:  symlink(): File exists in /var/www/openrepeater/includes/classes/Modules.php on line 353" while reading response header from upstream, client: 192.168.1.219, server: , request: "POST /modules.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "192.168.1.203", referrer: "https://192.168.1.203/modules.php"
2019/05/18 02:57:51 [error] 652#652: *947 FastCGI sent in stderr: "PHP message: PHP Warning:  symlink(): File exists in /var/www/openrepeater/includes/classes/Modules.php on line 353" while reading response header from upstream, client: 192.168.1.219, server: , request: "GET /modules.php?activate=74 HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "192.168.1.203", referrer: "https://192.168.1.203/modules.php"
2019/05/17 22:07:21 [info] 599#599: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/18 23:02:38 [info] 548#548: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/30 21:43:13 [info] 550#550: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/05/30 21:43:12 [info] 557#557: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/27 23:34:31 [info] 597#597: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/27 23:35:59 [error] 705#705: *3 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.106, server: , request: "GET /wizard/index.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.108", referrer: "https://192.168.1.108/login.php"
2019/07/27 23:35:59 [error] 705#705: *3 open() "/var/www/openrepeater/favicon.ico" failed (2: No such file or directory), client: 192.168.1.106, server: , request: "GET /favicon.ico HTTP/1.1", host: "192.168.1.108", referrer: "https://192.168.1.108/wizard/index.php"
2019/07/27 23:39:16 [info] 615#615: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/27 23:41:20 [error] 711#711: *31 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.106, server: , request: "GET /wizard/index.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.102", referrer: "https://192.168.1.102/login.php"
2019/07/27 23:44:43 [info] 564#564: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/27 23:52:05 [info] 553#553: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/28 00:13:05 [info] 578#578: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/28 00:21:39 [info] 577#577: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/28 00:34:27 [info] 605#605: Using 32768KiB of shared memory for nchan in /etc/nginx/nginx.conf:64
2019/07/28 00:43:26 [error] 773#773: *57 open() "/var/www/openrepeater/favicon.ico" failed (2: No such file or directory), client: 192.168.1.106, server: , request: "GET /favicon.ico HTTP/1.1", host: "192.168.1.102"
2019/07/28 01:12:25 [error] 775#775: *104 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.106, server: , request: "GET /wizard/index.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock", host: "192.168.1.102", referrer: "https://192.168.1.102/"

CR7AVR, 2019-07-29:
Problem Solved
Was a ip´s conflict, jut put static ip and now it works perfectly.

just edit "/etc/dhcpcd.conf" and difine static ip adress

thanks for everything

best regards
CR7AVR

alleghenycreative, 2019-07-30:
Thanks for the update and glad it is working now.

closed 2019-07-30 by alleghenycreative