OpenRepeater/openrepeater #80: Mumble Module

s3m1s0n1c, 2020-05-24:
Hey guys, 

any chance of getting a module to interface with mumble?

Mumble is a Low Bandwidth VOIP system which allows you to have rooms within the server and ability to lock it down so only authorized people can access the room.

Would be awesome to use mumble to link into a commercial repeater for those who don't have RF into the repeater but have mobile coverage.

Thanks
Sonic

ChZeman, 2020-09-13:
I'll second that!

alleghenycreative, 2020-09-14:
Certainly not opposed to it...it would just take someone familiar with svxlink and mumble to code the base svxlink module. Then it could be wrapped for ORP with a UI. 

There are other VOIP solutions, echolink is already supported and we have a beta module for FRN, Free Radio Network (as someone had already done this for SVXLink). SVXLink has a voip linking service as well but that has not been implemented yet. We are working on some new UI changes that should make that integration a bit easier. 

I can certainly add it to the feature requests.

alleghenycreative, 2020-09-14:
Also not being familiar with Mumble, security and compliance would be of concern. You don't want a non-licensed user being able to connect into an RF enabled system and making possible illegal transmissions. The advantage to something like EchoLink is the users must be validated first as being licensed. Just a consideration.

s3m1s0n1c, 2020-09-14:
**Hey,
Just a few thoughts on your statements**

> There are other VOIP solutions, echolink is already supported and we have a beta module for FRN, Free Radio Network (as someone had already done this for SVXLink). SVXLink has a voip linking service [...]
**Why limit this project to echolink or FRN?**
**Mumble is a great low bandwidth VOIP software which has better functions then FRN.**
**Check out [TalkiePi ](https://github.com/dchote/talkiepi)or [talkkconnect](https://github.com/talkkonnect/talkkonnect) they might give ya a bit more understanding of how to link**

> I can certainly add it to the feature requests.

**Sounds Good.**

> Also not being familiar with Mumble, security and compliance would be of concern. 

**Mumble is opensource and kept updated by a team of developers and is no difference to FRN or any other opensource software.**
**Mumble is in two parts.. A Server and The Client**
**Basicly you would want the client to connect to the server and link in with openrepeater via the "Link" function to feed audio both from the client to the repeater and from the repeater to the client**

You don't want a non-licensed user being able to connect into an RF enabled system and making possible illegal transmissions. The advantage to something like EchoLink is the users must be validated first as being licensed. Just a consideration.

**Mumble can be locked down by the owner of the mumble server so that only authorized users can access the room that has access to the link.**

**OpenRepeater could be used for any repeater being commercial or private not just Ham so authentication and "Unauthorized Users" depends on the application being used.**

**I guess time will tell and we'll see where this goes.**

**Thanks for your consideration**