OpenRepeater/openrepeater #81: PI-REPEATER-8x - New gpio settings

Dloranger, 2020-11-24:
Need to update the database to revise the default pins.  This is due to the addition of a relay board which caused pin offsets

COS1=446
COS2=447
COS3=454
COS4=455
COS5=462
COS6=463
COS7=470
COS8=471

PTT1=442
PTT2=443
PTT3=450
PTT4=451
PTT5=458
PTT6=459
PTT7=466
PTT8=467

Relay Board
Relay K1 = 488 (Active high)
Relay K2 = 489 (Active high)
Relay K3 = 490 (Active high)
Relay K4 = 491 (Active high)
LED 1 = 496 (Active low)
...
LED 7 = 503 (Active low)

jcacioppo59, 2020-12-03:
In addition,

The Relay Module in use at present uses the following GPIO's

Relay K1 = 480 (Active High)
Relay K2 = 481 (Active High)
Relay K3 = 482 (Active High)
Relay K4 = 483 (Active High)
Relay K5 = 484 (Active High)
Relay K6 = 485 (Active High)
Relay K7 = 486 (Active High)
Relay K8 = 487 (Active High)

Dloranger, 2020-12-03:
> In addition,
>
> The Relay Module in use at present uses the following GPIO's
>
> Relay K1 = 480 (Active High)
> Relay K2 = 481 (Active High)
> Relay K3 = 482 (Active High)
> Relay K4 = 483 (Active High)
> Relay [...]

This is for an off the shelf relay board that requires a few extra entries in the /boot/config.txt to instantiate, but maybe it could be a drop menu option like the wizard has for preset values.