OpenRepeater/openrepeater #85: ORP 2.2.0 Bug report Incomplete Backup of files

jcacioppo59, 2020-12-03:
During the system backup of 2.2.0 no files are created to recreate the "alsamixer" settings

alleghenycreative, 2020-12-03:
Thanks Jeff for picking up on this. The feature should be there, but I have briefly verified myself on one of my own backups that the alsa settings file doesn't exist. Again for others to look into the backup file, one would simply download the .orp file and change the extension to .tar.gz and use an appropriate program to unpack the archive. For alsa settings, the .orp package should contain a file called "alsamixer.state" in the root folder. 

The code is there in /var/www/openrepeater/includes/classes/BackupRestore.php on line 69 (2.2.x branch). I've manually tested the bash command from terminal and that seems to work. You could issue something as root user like "orp_helper alsa backup /tmp/alsamixer.state" and that should create a backup file in the /tmp folder. I have to dig into this more and see why it is not executing this statement from within the class. It worked when originally implemented.

alleghenycreative, 2020-12-04:
This bug/issues has been squashed. It was fixed in commit 84202eb to 2.2.x branch. While there has been a number of white space changes, the meat of the fix is at line 238 (previously 237) in the condition of the if statement. It was 
`if (file_exists($this->alsa_state_file)) {`
and has changed to
`if ( file_exists($this->archive_build_dir . $this->alsa_state_file) ) {`

It was basically checking to see if the file existing, but since it was not pointing to the full path to the file, it was not adding it to archive. I've tested this on my local machine and it is now present when unpacking the .orp file download from the controller.

closed 2020-12-04 by alleghenycreative