OpenRepeater/openrepeater #86: No output sound when building 2.2.x from script on fresh install in Raspbian Lite and ICS v1

yuretszzz, 2020-12-08:
I have working repeater running 2.1.2.
Because of audio delay on TX I want to update svxlink to newer version for adding support of AUDIO_DEV_KEEP_OPEN=1.
But I can get output sound on 2.2.x. All were configured as on 2.1.2 but still no luck.
Temporary solution I compiled svxlink from git and updated it in 2.1.2, TX delay disappeared,  but because  of updates my old svxlink.conf don't work very well. For example I cant disable SHORt_ID at all.
My question is how can I get updated version of ORP and svxlink? 
Thank you.

alleghenycreative, 2020-12-09:
Hello @yuretszzz,
Yes the script for 2.2.x uses the last release of SVXLink (Sept 2019). Which I believe is the release that added the support for AUDIO_DEV_KEEP_OPEN. There is also another ASYNC_AUDIO_ALSA_ZEROFILL that we've tested along with that. 2.2.x also has a dev UI that allows you to add these variable using an hidden form. The plan is to incorporate that into the new 3.0 UI design. 

My concern would be to find out why you are not getting audio in 2.2.x build. First thing that comes to mind would be alsamixer settings. Another thing might be the version of Raspbian/Raspberry Pi OS that you are using. I haven't done a new build myself since they've renamed Raspbian to Raspberry Pi OS. In theory it should be fine since it is still Debian 10 (buster) under the hood. I did see just the other day Raspberry Pi OS is suppose to be making pulse audio it de facto standard. I don't know if this will apply to the lite version. What version of Raspberry Pi OS did you use?

I am just starting a new dev build using the Raspberry Pi OS release Dec 4th, 2020. I wanted to start with a clean slate and also test the new boot loader from a USB drive. No issues there it works fine with USB on my Pi 4. I ran the 2.2.x script but installed as a dev install instead of the express/standard install. I have not been able to hook it up to my test rig yet and check for actual audio output. But remoting into it everything else looks normal. 

ORP does some custom TCL logic as well. So in writes out some different sections for the output identifications. So that might be what you are experiencing with 2.1.2 and trying to disable the short ID. It all depends on where you are placing the code.

yuretszzz, 2020-12-09:
How can I access to dev UI?

alleghenycreative, 2020-12-09:
It's only in the 2.2.x build and it's hidden from the regular UI for the purpose of developers and testers. It's easy to access. You will need to log in first (i.e. in another tab) to establish the sign in session. Then in another browser tab go to https://x.x.x.x/dev_ui/, replacing the x.x.x.x with the IP of your Pi. That will bring you up to index page with with a list of different options.  Advanced SVXLink Settings would be what you'd want to add in the AUDIO_DEV_KEEP_OPEN and  ASYNC_AUDIO_ALSA_ZEROFILL settings.

yuretszzz, 2020-12-09:
Please note that same hardware setup works well with 2.1.2 from IMG. 
But on 2.2.x from script no sound output.
Very odd for me that capture works, if I transmit vi RF, I can hear my self on Echolink. 
My setup:
downloaded and installed latest Raspbian Lite:
Linux openrepeater 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
root@openrepeater:~# amixer
Simple mixer control 'Headphone',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 0 [0%] [-51.50dB] [off]
  Front Right: Playback 0 [0%] [-51.50dB] [off]
Simple mixer control 'Headphone Mux',0
  Capabilities: enum
  Items: 'DAC' 'LINE_IN'
  Item0: 'DAC'
Simple mixer control 'Headphone Playback ZC',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'PCM',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 192
  Mono:
  Front Left: Playback 154 [80%]
  Front Right: Playback 154 [80%]
Simple mixer control 'Lineout',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 31
  Mono:
  Front Left: Playback 25 [81%] [-3.00dB] [off]
  Front Right: Playback 25 [81%] [-3.00dB] [off]
Simple mixer control 'Mic',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 3
  Mono: 0 [0%] [0.00dB]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 15
  Front Left: Capture 12 [80%] [on]
  Front Right: Capture 12 [80%] [on]
Simple mixer control 'Capture Attenuate Switch (-6dB)',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Capture Mux',0
  Capabilities: enum
  Items: 'MIC_IN' 'LINE_IN'
  Item0: 'LINE_IN'
Simple mixer control 'Capture ZC',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'AVC',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'AVC Hard Limiter',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'AVC Integrator Response',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 3
  Mono: 0 [0%]
Simple mixer control 'AVC Max Gain',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 2
  Mono: 0 [0%] [0.00dB]
Simple mixer control 'AVC Threshold',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 96
  Mono: 12 [12%] [12.00dB]
Simple mixer control 'BASS 0',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 95
  Mono: 1 [1%]
Simple mixer control 'BASS 1',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 95
  Mono: 0 [0%]
Simple mixer control 'BASS 2',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 95
  Mono: 0 [0%]
Simple mixer control 'BASS 3',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 95
  Mono: 0 [0%]
Simple mixer control 'BASS 4',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 95
  Mono: 0 [0%]
Simple mixer control 'DAP MIX Mux',0
  Capabilities: enum
  Items: 'ADC' 'I2S'
  Item0: 'ADC'
Simple mixer control 'DAP Main channel',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 65535
  Mono: 0 [0%]
Simple mixer control 'DAP Mix channel',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 65535
  Mono: 0 [0%]
Simple mixer control 'DAP Mux',0
  Capabilities: enum
  Items: 'ADC' 'I2S'
  Item0: 'ADC'
Simple mixer control 'Digital Input Mux',0
  Capabilities: enum
  Items: 'ADC' 'I2S' 'Rsvrd' 'DAP'
  Item0: 'I2S'
 Advanced Details

Audio Input Devices
**** List of CAPTURE Hardware Devices ****
card 0: Audio [Fe-Pi Audio], device 0: Fe-Pi HiFi sgtl5000-0 [Fe-Pi HiFi sgtl5000-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
Audio Output Devices
**** List of PLAYBACK Hardware Devices ****
card 0: Audio [Fe-Pi Audio], device 0: Fe-Pi HiFi sgtl5000-0 [Fe-Pi HiFi sgtl5000-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
Advanced Linux Sound Architecture Driver Version k5.4.79-v7+.

yuretszzz, 2020-12-09:
Problem solved

alleghenycreative, 2020-12-09:
Glad to Hear it. 

So you had a bad backup? I just discovered recently that the alsa backup file wasn't getting copied into the backup (.orp) file like it was supposed to. I just recently patched that in 2.2.x.

yuretszzz, 2020-12-09:
I don't restored system from backup. It was new clean install by 2.2.x script. And for some reason it don't work properly for also.

alleghenycreative, 2020-12-09:
I will take a look at my new build as soon as I can and see if I can get good results.

Dloranger, 2021-03-17:
amixer -c 0 set Lineout playback 80% unmute

They changed the behavior of the codec drivers, the lineout is muted by default now.  You need to unmute it to get audio on the line-out

alleghenycreative, 2021-03-19:
Yes we've been struggling with this for a while trying to build some test images in the last month or two. I can confirm that Dan's command above should correct this. I just have to figure out how to pass that into ALSA now as part of the board presets.

alleghenycreative, 2021-04-10:
This was patched by adding updated values to the board preset file. This will push the correct settings into ALSA and unmute the Fe-Pi codec.

2.2.x
https://github.com/OpenRepeater/openrepeater/commit/1546f342498b1f50070a5626f7be95e884c9079c

3.0.x
https://github.com/OpenRepeater/openrepeater/commit/c644751de647f57edd67d86c5f2a73c208c08078

closed 2021-04-10 by alleghenycreative