OpenRepeater/openrepeater #87: Custom ident is not playing, still only morse

lukasalex, 2020-12-29:
Hello guys,
I was testing custom identification. I tried mp3, wav, converted wav/mp3 by sox to 16bit16mhz according to manual but still repeater is using morse. 
Is there any trick or it's a bug?
I am using latest version on Raspberry Pi 3B
Thank you

alleghenycreative, 2021-01-02:
When you say you are using the latest version, what is the version number? The latest image version is 2.1.2, unless you are using the build script to create a 2.2.0 image which is development code as of this date. 

How are you uploading your sound files? Manually? You shouldn't need to do any SOX conversion on your end. Just upload a WAV or MP3 file via the upload button on the Identification page (when in custom mode) and select the file after uploaded. The upload process will take care of converting the audio in the background to the appropriate sampling rate of WAV file that SVXLink expects. 

You may need to refresh the page when changing modes. Sometimes the AJAX doesn't take. Be sure to rebuild and restart after the changes are made to apply them to the live system. Also be sure to check the Log tab, maybe open up in a separate browser tab. If you have bad audio files that cannot be played, you should see it there. I am suspecting if you converted the file yourself and it cannot play it via SVXLink and you have append morse ID option selected then that might be why you are only hearing the CW. The log should make that clear. Hope that helps.

lukasalex, 2021-01-02:
Hi Aaron, 
thank you for your inputs. 

I am running 2.1.2 official release, using USB audio dongle, 
[image]
Here is my config for identification:
[image]
After this I am restarting the service by red notification and checking logs, but as you said, somwhere will be mistake, it's not sending log about identification service, only sending log about morse long/short but it's 4 seconds long (mp3/wav is 7 seconds).
[image]
Here are sounds, before and after upload.
[sounds.zip](https://github.com/OpenRepeater/openrepeater/files/5760872/sounds.zip)

Thank you for any advice ;-)
Lukas

lukasalex, 2021-01-03:
Hi. I tried to upload again samples and set it as identification file, changed time so short is 1 minute, long is 10 minutes and still it's using morse.
[image]

[image]

[image]
[image]
[image]

lukasalex, 2021-01-03:
And here is backup file [jn89fc_2021-01-03_05-28-42.zip](https://github.com/OpenRepeater/openrepeater/files/5761211/jn89fc_2021-01-03_05-28-42.zip)

lukasalex, 2021-01-03:
Ahaaa, I find it: 
/usr/share/svxlink/events.d/local/Logic.tcl
here is 
if {$CFG_TYPE == "Repeater"} - I am running Simplex

so I changed it to if {"x"=="x"} and now it's sending wav file to TX :)
So if you can add Identification for Simplex too (or... I should let it with "x" ? 👍 
Lukas

lukasalex, 2021-01-03:
/var/www/openrepeater/includes/classes# vi ./SVXLink_TCL.php

Edited so now it's still active :-)

Lukas

alleghenycreative, 2021-01-05:
@lukasalex, Sorry for the delay. I was waiting until I had the opportunity to try and download your sound files and test for myself. I didn't notice your additional comments because of how my email grouped the replies. 

Glad to hear that you got it figured. out. I didn't know that you were running it in simplex mode. Yes as you discovered the CW ID is/was hard coded. The simplex node option was originally created as a work around to allow for simplex echolink node usage and I guess it was thought at the time that nothing more would be needed than CW. That part of the UI is in the process of being reworked so that should change in a future release.

lukasalex, 2021-01-08:
Thank you for your reply ;-) 

I have one additional question, so I will open it in new case ;-) 

Lukas

closed 2021-01-08 by lukasalex