OpenRepeater/openrepeater #88: DTMF code options

lukasalex, 2021-01-08:
Hi again, 

It will be nice to have opportunity to change or add function for DTMF 

I am using simplex node and I am looking for similar function as it is on other repeaters -
"*" start of the recording
"#" end of the recording
Perfect will be if I can code what will be start and end symbol

Right now is parrot activated for some time
1# activation
Notification about activation
Communication (without any DTMF tone)
.....time (or "#" for end of the module)
End of the parrot 
Notification about deactivation
If this is hard to change, nice option will be to have opportunity (switch button) to mute the de/activation. 

Thank you ;-)
Lukas

Dloranger, 2021-01-08:
Svxlink does provide the option to change/customize/add the dtmf codes to some extent.  However there is very little documentation about this.  Search online for modifying the svxlink tcl files such as logic.tcl and locale.tcl.

lukasalex, 2021-01-09:
Thanks man, I found a way, how to mute it, so it's working perfectly. Right now need to go through DTMF, need to change 
Identify = "*" to "#"
Start parrot = "1#" to "*"
And then we are done :-)
Any advice with the file/location will be appreciated. 
Lukas

Dloranger, 2021-01-09:
Parrot setting are in the module config files, /usr/share/svxlink/... or
/etc/svxlink/... if you don't want parrot at all, you can just remove it
from the svxlink.conf file settings.

# is a global system symbol, you won't be able to change it without a lot
of headache and changing source code, recompiling, etc.

If you want to force ID, you can just use the star dtmf instead which is
the default.  I believe this can be changed, but I forget which setting it
is. The star, pound and letters all have special meanings in svxlink so you
will have to use them as intended

alleghenycreative, 2021-01-10:
Both * and # are reserved characters in SVXLink. * forces identification/status and # follows most commands and by itself is used to disable modules. 

SVXLink does support macros. This is not supported by the current release of ORP 2.1.2. The 2.2.x development branch does have some back end code that is available for testing to add the macros configuration. So something like D1# could enable say the Echolink module and connect to a specified node. Then D2# could be used to disconnect from the node and disable the module. You could interact with other modules the same way.

lukasalex, 2021-01-10:
Aha, nice info. And it's possible to do
D5 to activate parrot?
Target is to activate parrot with one DTMF symbol and end it with another so change will be 

Activate parrot DTMF  "5"
Deactivate parrot DTMF "#"

I know that parrot can be activated once and communication can be without any DTMF codes. Target is that operator has to activate it every time.
Lukas