OpenRepeater/openrepeater #108: Unable to select single record from database

tmastersmart, 2023-03-23:
New install. After logging in I get

Unable to select single record from database

abcrawford, 2023-03-24:
I’ve seen this occur before when the SD card image hasn’t been expanded to fill the card. You can use the raspi-config utility on Raspberry Pi OS to do this.

legacycode, 2024-01-13:
I have the same issue. I copied the 2.1.3 version to sdcard and wait until it boots up. I tried to login into the dashboard and get the same error message. The sdcards seems to be expanded correctly ba openrepeater. I am using a raspberry pi model 4:

root@openrepeater:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30388284 2302700  26823852   8% /
devtmpfs          823564       0    823564   0% /dev
tmpfs             955660       0    955660   0% /dev/shm
tmpfs             955660   16824    938836   2% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             955660       0    955660   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   49718    208378  20% /boot
tmpfs             191132       0    191132   0% /run/user/0

legacycode, 2024-01-13:
Ok issue solved. I used the wrong user and password.

Username: admin
Password: openrepeater

Sorry and best regards.